bambulab/BambuStudio #6748: Printer not found in studio, but Android app is working

kawogi, 2025-04-29:
### Bambu Studio Version

2.0.3 (flathub), 2.0.3.54 (built from source), 1.10.1.50

### Where is the application from?

Bambu Lab github releases

### OS version

Linux Mint 22

### Printer

BambuLab X1-Carbon

### How to reproduce

1. switch on printer
2. log in to account using the Android app (printer confirms log-in and shows my account)
3. install BambuLab-studio
4. confirm installation of network module
5. confirm sync with cloud
6. log in to account (verified by PIN sent via mail; region set correctly)
7. switch to Device-Tab (shows "No Printer +")
8. click on "No Printer +" (options shown: "Bind with Pin Code", "Bind with Access Code")
9. On Printer: Test Network → Router 8 ms / Internet 11 ms
10. ping the printer from the PC → success

### Actual results

The Printer can be seen and controlled from the Android App

In the studio the Printer isn't shown under "My Devices".

Adding a printer via "Bind with Pin Code" isn't feasible because the PIN seems to be unobtainable. The Printer's Account page only shows the region, my avatar and name

Adding a printer via "Bind with Access Code" lets me fill out all fields and after confirmation, the dialog closes without any message or change

### Expected results

When clicking on "My Devices" the printer, connected to my account shall be shown.

When using "Bind with Pin Code" the required Pin Code shall be obtainable from somewhere

When using "Bind with Access Code" the Printer shall be added to the list after completing the last dialog or an appropriate error message shall be shown.

### Project file & Debug log uploads

[log and project.zip](https://github.com/user-attachments/files/19951123/log.and.project.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

walterwongbbl, 2025-05-07:
Hi. @kawogi  Bind with Pin Code Only supports P-series printers.
If you switch the printer to LAN-ONLY mode and then directly select the printer and connect it in the ordinary way, will it be successful?

kawogi, 2025-05-10:
(update: I installed the latest firmware for the printer (01.09.00.01) and AMS (00.01.06.62))
While being logged in to my account to BambuStudio (fresh install from flathub v2.0.3):

- switched printer into LAN-Mode
- clicked on "Device" tab (no printers listed)
- click on "No Printer +" → "Bind with Access Code" (dialog opens)
- enter IP (verified by pinging), enter access code and confirm → shows message "connecting printer … dialog will close …" and then the dialog disappears. No printer was added no further message

I can reproduce this procedure multiple times. I tried restarting the studio and also renewed the access code. Right now there's not way for me to reach the printer.

kawogi, 2025-05-10:
When installing the studio I'm being asked whether I want to install some "network plug-in(?)" which I confirmed.
Does my problem description match a malfunctioning network-module? Is there a way to -re-install this module or test it?

I found a built-in network test which looks ok to me:

```
Sat May 10 02:33:10:start_test_sequence
Sat May 10 02:33:10:test bing start...
Sat May 10 02:33:10:[test_bing]: url=http://www.bing.com/
Sat May 10 02:33:10:test bing ip resolved = 2.17.22.34
Sat May 10 02:33:10:test bing ok
Sat May 10 02:33:10:test bambulab start...
Sat May 10 02:33:10:[test_bambulab]: url=https://api.bambulab.com/v1/iot-service/api/slicer/resource?name=slicer&version=02.00.03.54&guide_version=00.00.00.01
Sat May 10 02:33:10:test bambulab ip resolved = 104.18.35.41
Sat May 10 02:33:10:test bambulab ok
Sat May 10 02:33:10:test storage start...
Sat May 10 02:33:10:[test_oss]: url=http://upload-file.bambulab.com
Sat May 10 02:33:10:test storage failed
Sat May 10 02:33:10:status=404, body={"error_msg":"404 Route Not Found"}
, error=
Sat May 10 02:33:10:test storage upgrade start...
Sat May 10 02:33:10:[test_oss_upgrade]: url=http://upgrade-file.bambulab.com
Sat May 10 02:33:11:test storage upgrade failed
Sat May 10 02:33:11:status=404, body={"error_msg":"404 Route Not Found"}
, error=
Sat May 10 02:33:11:test storage download start...
Sat May 10 02:33:11:[test_oss_download]: url=https://api.bambulab.com/v1/iot-service/api/slicer/resource?slicer/plugins/cloud=02.00.03.00
Sat May 10 02:33:12:test storage download ok
Sat May 10 02:33:12:test plugin download start...
Sat May 10 02:33:12:[test_plugin_download]: downloadurl=https://public-cdn.bblmw.com/upgrade/studio/plugins/02.00.03.55/7cfa549ef4/linux_02.00.03.55.zip
Sat May 10 02:33:14:test plugin download ok
Sat May 10 02:33:14:end_test_sequence
```