bambulab/BambuStudio #6749: Keep Crashing while slicing.

kingdomheartstao, 2025-04-29:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab github releases

### OS version

W10

### Additional system information

CPU: Intel Xeon W-2275
Memory: 128G
GPU: RTX A5000

### Printer

H2D

### How to reproduce

Open any 3mf on makerword, sync then slice. when the bar show it complete all layers.

### Actual results

Crash

### Expected results

not crash

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/19954418/log.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

kingdomheartstao, 2025-04-29:
My other AMD PC works fine tho.

BoBoYu-BamBu, 2025-04-29:
@kingdomheartstao Sorry, could you provide me with the 3mf file?

XunZhangBambu, 2025-05-15:
@kingdomheartstao Can you refer to https://forum.bambulab.com/t/bambu-studio-crashes-after-slicing-solved-nvidia-problem/162392 and give it a try?