bambulab/BambuStudio #6750: Ctrl-D not working in version 2.0 +

tvixen, 2025-04-29:
### Bambu Studio Version

version 2.0 and rest of this versions

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

CPU Inter I9
64Gb RAM

### Printer

A1 Mini and P1S

### How to reproduce

In the software when a model is on the plate.
Normal user can press Ctrl-D to delete the whole plate, but after version 2.0 this is not possible.

### Actual results

Nothing happens when pressing Ctrl-D

### Expected results

The whole plate is deleted

### Project file & Debug log uploads

All .STL files

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2025-04-30:
Hello, we have changed the latest shortcut key to ctrl+shift+d

[image]

closed 2025-04-30 by Haidiye00

tvixen, 2025-06-02:
But why make a change to the general windows delete key combination?   Everybody else use this combination!

Haidiye00, 2025-06-03:
Hello, considering the Mac, I made this modification. Sorry for the inconvenience. Actually, I usually select all with Ctrl+A and then press delete.