bambulab/BambuStudio #6754: Bambustudio crashes when loading the attached simple step file.

Panda631, 2025-04-29:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Win11

### Additional system information

  

### Printer

X1C

### How to reproduce

Import the attached step file into BS.
The program hangs...

The file can be opend correct in any other 3D-render applikations (e.g. Mayo)

[Test.zip](https://github.com/user-attachments/files/19962012/Test.zip)

### Actual results

Ther step file can't be opened.

### Expected results

Ther step file should be opened without problems

### Project file & Debug log uploads

No file

### Checklist of files to include

- [ ] Log file
- [ ] Project file

JohnSmithCYM, 2025-04-30:
[image]

I am not sure the parameters you use. You can use my parameters like above picture and give it a try. Everything works fine here.

[image]

Panda631, 2025-04-30:
As mentioned above. I can't open.

MackBambu, 2025-04-30:
@Panda631 You need to provide the file import parameters, andhardware information, including CPU and memory. Additionally, could you try importing a simple cube STEP file?

Panda631, 2025-04-30:
@MackBambu 
Any other files load flawless in seconds. It seems to depend on that special shape of the body?

What do you mean by import parameters?
I just start BS -> [create new project] -> [Add (ctrl-I)] -> [select test.step] 
The open Dialog apears and stays for minutes......

[image]

Here 3mf file:

[Test-3mf.zip](https://github.com/user-attachments/files/19975183/Test-3mf.zip)

My HW:
AMD Ryzan7 3800X 8-core @3,89GHz
NVIDIA GeForce GTX1060 6GB
16BG RAM
Win11 Pro

MackBambu, 2025-04-30:
>What do you mean by import parameters?

When you import the file, does this dialog appear? If not, please try enabling this option and increasing its parameter value.

[image]

[image]

Panda631, 2025-04-30:
@MackBambu 

This dialoge was disabled.
Linear deflection was set to 0.001

I followed your proposal and incread it to 0.03 and it loads imediatelly.
Many thanks, that was the solution. Great job!

closed 2025-04-30 by Panda631