bambulab/BambuStudio #6755: Fatal Error Access Violation!

Orel-A, 2025-04-29:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Pro 24H2

### Printer

Bambu Lab A1

### How to reproduce

Click slice on the provided 3mf file

### Actual results

Can't slice. Fatal Error!

### Expected results

To slice

### Project file & Debug log uploads

[Egg.zip](https://github.com/user-attachments/files/19963743/Egg.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

JohnSmithCYM, 2025-04-30:
Thanks for your feedback, question is confirmed and we are analyze it.

zhimin-zeng-bambulab, 2025-05-13:
[Egg-modified.zip](https://github.com/user-attachments/files/20188328/Egg-modified.zip)

There is a problem with your file. We have uploaded the repaired file here. How did you generate this file?

Orel-A, 2025-05-13:
> How did you generate this file?

I was trying to change the layer height modifier's range and the bug occured

zhimin-zeng-bambulab, 2025-05-16:
@Orel-A  Could you reproduce the problem that produces the wrong file?

Orel-A, 2025-05-21:
Unfortunately, I was not able to reproduce the error.

closed 2025-05-21 by Orel-A