bambulab/BambuStudio #6756: Bottom Surface

JJ-1958, 2025-04-29:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Pro Version	10.0.26100 Build 26100

### Printer

Bambu Lab X1C

### How to reproduce

Change bottom surface pattern

### Actual results

No effect no matter what pattern is selected

### Expected results

Pattern remains at default

### Project file & Debug log uploads

[Cube.zip](https://github.com/user-attachments/files/19964118/Cube.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

JohnSmithCYM, 2025-04-30:
[image]
Here is my result and the 3mf is here.Everyting works fine like the picture, can you give a screenshot or gif to show the process?
[BottomToArchi.zip](https://github.com/user-attachments/files/19970328/BottomToArchi.zip)

JJ-1958, 2025-04-30:
The cube works fine.  Initially I thought I could illustrate the
problem with the cube.  Here is the project part I am working on now.  What
I would really like to see is the bottom surface as monotonic with 1 wall.
In Bambu Studio you can specify Wall# and surface pattern.

Anyway, I can't get the bottom surface of this lever to change pattern.
Thanks for looking into this for me

JohnSmithCYM, 2025-05-06:
Thanks for your feedback, and I don't know if something else bother you. Can you show the your problem with a picture or close the issue if you this doesn't bother you anymore?

closed 2025-05-08 by BoBoYu-BamBu