bambulab/BambuStudio #6757: New Bambu Studio will not generate normal supports any more. Auto nor Manual

jaytrenell, 2025-04-29:
### Bambu Studio Version

Bambu Studio 2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

H2D

### How to reproduce

When generating a print I have been printing over a year. Upgraded to 2.0 with my addition of H2D.  Sliced file as normal.  No normal supports will generate. Only the bottom brim support surface. No supports (normal) then the object is there unsupported.  Normal Supports (auto or painted manual does not produce supports once generated.  Do not want to add 2 hours to print to use tree supports. 

[image]

### Actual results

No middle supports were generated thus print fails even though normal supports are enabled.  Picture of Bambu Studio 2.0. 

[image]

### Expected results

[image]

Picture from gcode generated Bambu Studio 1.x . Same file. 

### Project file & Debug log uploads

[rykersidegrill2024 v3.zip](https://github.com/user-attachments/files/19965507/rykersidegrill2024.v3.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

NothFie, 2025-05-06:
@jaytrenell Thanks for your feedback. It's because the LIGHTNING Infill does not support NORMAL support. If you change the base pattern, the supports generate well. It's indeed a BUG and we will fix it in the future versions.

NothFie, 2025-06-03:
@jaytrenell This issue should be fixed in the latest version Studio V2.1.0, please have a check.

fahaaaxiki, 2025-07-21:
Hello, we have solved the problem at latest release. please check [v02.02.00.60](https://github.com/bambulab/BambuStudio/releases/tag/v02.02.00.60)