bambulab/BambuStudio #6759: Assemble - Point and Point Assembly, does nothing after points selected, on an Assembly. No options to move objects together.

jwmann, 2025-04-29:
### Bambu Studio Version

Version 2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS 15.4 (Sequoia)

### Additional system information

MabBook Pro M1 Max

### Printer

Bambu Lab A1 Mini

### How to reproduce

https://github.com/user-attachments/assets/908b2ac4-ec89-492b-91b6-8b6ee33f5ebf

1. Open New Project
2. Add Primitive Cube, Move to the Left
3. Add Primitive Cube, Move to the Right
4. Select both cubes, Right-Click, Merge
5. Select new Assembly, Select the Tool 'Assemble'
6. Choose Option 'Point and Point Assembly' from the Dropdown.
7. Select Two opposing points on separate cubes

### Actual results

Assemble Tool shows the distance between the points, but no option to move them together

### Expected results

An option to move the points together, similar to the Face and Face Assembly option

### Project file & Debug log uploads

No Project or Log File is Needed

### Checklist of files to include

- [ ] Log file
- [ ] Project file

jwmann, 2025-05-03:
Clicking 'Return' on the top left after selecting the points just resets everything. (AKA, nothing happens)

Hurly26, 2025-11-26:
I don't think the instructions for this tool are very clear and instructions are in multiple locations. See wiki links below. The summary is, to move the parts, go into each X, Y, and Z box and adjust the values to what you want them to be. After clicking the two points, the deltas are shown. If you want the points to be coincident instead, enter X = 0, Y = 0, Z = 0. 

This link shows how to use the tool fairly well: https://wiki.bambulab.com/en/software/bambu-studio/assemble

This is a related link but I don't think it explains it well enough: https://wiki.bambulab.com/en/bambu-studio/assemble-tool

Note that I have discovered an apparent bug with this tool. If your Bambu Studio "Preferences" have "Units" set to imperial, updating the X, Y, Z values will not work as intended on 2.4.0.70. It appears the tool only works with metric units at this time. If you use the tool and come up with an X value of 0.43 inches for instance and update this to say 0, the model will actually move by (0-0.43) mm = -0.01 inches and the new X value will be 0.41 instead of the expected value of 0. This can further be confirmed by taking the reported value in inches, multiplying by 25.4 to convert to mm and then subtracting that from the reported value. This will place the model at a coincident X.