bambulab/BambuStudio #6761: Supports fails using all bed

JoaquinFB, 2025-04-29:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambu Lab X1 Carbon

### How to reproduce

[image]

Using all bed with a piece that need supports.
Excluded Area: 0x0, 18x0, 0x0, 0x0

### Actual results

Supports fails now

### Expected results

In previews releases this works! and supports was created in the bed with thin tree support.

### Project file & Debug log uploads

[Projectfailsupport.zip](https://github.com/user-attachments/files/19967170/Projectfailsupport.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

JoaquinFB, 2025-04-29:
I add a video, from the previus versions from bambu where you can see that the supports was right!

JoaquinFB, 2025-04-29:
And this supports if you remove the option: support_remove_small_overhang, create strange supports //middle inside the piece, middle in the bed but without connection.

[image]

NothFie, 2025-05-06:
@JoaquinFB Thanks for your feedback. The Organic Tree with "remove small overhang" closed might meet your needs.
The bottom surface is too small so it's classified into "small overhang", which is clearly wrong with eyes. We will keep improving the detection of overhangs to fix it in future versions.
The Slim Tree keeps growing radius to make it stable, sometimes when tree  is splitted into several pieces, the "threading" bug in your picture 2 appears. Also we are trying to fix it.

jfuentesb, 2025-05-17:
Gracias!