bambulab/BambuStudio #6767: Floating Tree Support

shuairenchen, 2025-04-30:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Version 24H2 (OS Build 26100.3775)

### Printer

P1S with AMS Combo

### How to reproduce

Slicing the "Alita bust" model from Fotis Mint, using the default "0.12mm High Quality @BBL X1C" profile, with Tree support enabled.

### Actual results

Support generated in mid-air, causing print to fail.

### Expected results

Support is generated with the root on the print bed instead.

### Project file & Debug log uploads

[Alita Bust](https://drive.google.com/file/d/1yNM75szZTXG5uxA-Dj6dVhaGY-IHY1pJ/view?usp=sharing)

### Checklist of files to include

- [ ] Log file
- [x] Project file

shuairenchen, 2025-04-30:
[image]

NothFie, 2025-05-06:
@shuairenchen Thanks for your feedback. Sorry but I cannot reproduce your result in the 2.0.3.54 version Bambu Studio. Have you changed any other parameter? If you are indeed using the latest version of Studio and haven't changed other settings, please provide the sliced file(.gcode.3mf) to help me further analyze the issue.
[image]