bambulab/BambuStudio #6769: Polymaker PA6-GF missing from Studio

Shaische, 2025-04-30:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Sequoia 15.4.1

### Additional system information

On my X1 printer, I am able to specify Polymaker Fiberon PA6-GF, but as soon as I sync it from Bambu Studio, I get an error that the filament isn't "found" and I need to fake it as a different filament. Why is this filament available to choose from on the printer but not Bambu Studio?

### Printer

X1

### How to reproduce

Choose Polymaker Fiberon PA6-GF from filament list on the printer.
Sync it to Bambu to get error, it won't allow to add the filament.

### Actual results

It can't add it.

### Expected results

It should be in Bambu Studio.

### Project file & Debug log uploads

None

### Checklist of files to include

- [ ] Log file
- [ ] Project file

XunZhangBambu, 2025-05-10:
@Shaische  thanks. we will fix it later

XunZhangBambu, 2025-06-03:
@Shaische Hi, this has been fixed in our latest version 02.01.00.59