bambulab/BambuStudio #6772: Cannot access "Edit Process Settings" – stuck on "Switch to per-object setting mode" message

matcha1312, 2025-04-30:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambulab A1

### How to reproduce

1. Go to "Prepare"
2. Right click object
3. select "Edit Process Settings"
4. error message : "Switch to per-object setting mode to edit process settings of selected objects" with green arrow pointing to the "objects" tab in the sidebar
5. message appears even when I'm already in per-object setting mode, and even if I right-click directly from the sidebar.

### Actual results

error message : "Switch to per-object setting mode to edit process settings of selected objects" with green arrow pointing to the "objects" tab in the sidebar

### Expected results

no error message

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/19975613/log.zip)

[Ultimate+Filament+Spool+Enclosure+for+AMS+Lite-241111(2).zip](https://github.com/user-attachments/files/19975619/Ultimate%2BFilament%2BSpool%2BEnclosure%2Bfor%2BAMS%2BLite-241111.2.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

Haidiye00, 2025-05-06:
Hello, can you download ScreenToGif software and record a GIF of the operation video and upload it? Thank you
[Release ScreenToGif 2.41.1 · NickeManarin/ScreenToGif](https://github.com/NickeManarin/ScreenToGif/releases/tag/2.41.1)

jmackraz, 2025-07-29:
I'm experiencing this (V 2.1.1.52).  Here's a screen shot after selecting Edit Process from right-click menu on selected object.  I hope you do not need more screen capture.

[image]

Absoblogginlutely, 2026-02-03:
same issue 2.5.0.64
I'm in object mode, right click edit process settings and get this message.
The screen even highlights with a flashing arrow that the screen is already in object mode.

I've seen references to toggling from simple to expert mode but I don't see this option in the software anywhere either.
Enabling developer mode did not fix the error message popping up or enable an expert mode.

fanblade, 2026-03-09:
same issue 2.5.0.66

emelier, 2026-09-24:
I started encountering this issue too. Anyone come up with a solution?

dn768, 2026-10-14:
This is absolutely a bug!! It is, however, a bug in OrcaSlicer and not only in BambuStudio.

dn768, 2026-10-14:
> 17\. I started encountering this issue too. Anyone come up with a solution?

You just have to modify the settings in the object pane like normal. The menu item isn't designed to do anything 🤦‍♂️