bambulab/BambuStudio #6774: Printer setting "Auxiliary part cooling fan" doesn't turn off the aux fan

imagitech01, 2025-04-30:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab P1S

### How to reproduce

[image]

Uncheck the "Auxiliary part cooling fan" setting

### Actual results

Slicer still uses the Auxiliary part cooling fan speed setting from the Filament settings

[image]

### Expected results

If the checkbox is turned off, the aux fan should stay off.

### Project file & Debug log uploads

-

### Checklist of files to include

- [ ] Log file
- [ ] Project file

zhimin-zeng-bambulab, 2025-05-10:
Thanks for reporting this issue, we will check it.

QingZhangBambu, 2025-07-21:
it had been fixed in the latest version, please download and try

benvanmierloo, 2025-07-26:
confirmed solved! Thanks!