bambulab/BambuStudio #6775: PLC-CF filament with hardened steel extruder not able to specify in bambulab

etlaare, 2025-04-30:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

13th Gen Intel(R) Core(TM) i9-13900KF   3.00 GHz
32,0 GB 

### Printer

Bambu Lab P1S

### How to reproduce

1 select a model
2 change filament in printer to PLA_CF (it has the hardened steel option installed)
3 select generic PLA-CF
4 Slice
5 select Print
6 refuses to print, exclamation mark says selct filament manually it does not give any option and says I should change to PLA, in the device settings it also is not possible to change the filament, it still says PLA there but no other options available. The old option with confirming I had the correct hotend worked fine but now I can nto print with CF and wood anymore. Please prioritize, I need this print very urtently 

### Actual results

[image]
[image]

### Expected results

manual option should be available or in device it should be possible to change to PLA-CF. Which is not possible so the print buitton never goes green.

### Project file & Debug log uploads

[problem record.zip](https://github.com/user-attachments/files/19978968/problem.record.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

BoBoYu-BamBu, 2025-05-06:
Sorry, we are currently addressing this issue. You can first manually set the extrusion consumables to PLC-CF in the device page, then send the print job. It should be possible to print with the extrusion materials now.

etlaare, 2025-07-03:
Just installed 2.1.1.52 The problem still exists, it does not allow me to change the material on the device page and refuses to print from the external spool. Maybe this problem does not occur with an AMS but I don't have one. The only solution is still to downgrade to the version that asks you to verify there is a hardened steel nozzle.

XinZhangBBL, 2025-07-04:
@etlaare Could you please reproduce and video it.  So that we can fix the probelm.

> Just installed 2.1.1.52 The problem still exists, it does not allow me to change the material on the device page and refuses to print from the external spool. Maybe this problem does not occur with [...]

layooesh, 2025-07-28:
I have the AMS and does not let me print, select it. In the past it did well.
It recognises it but only lets me chose spool A3.

[image]

XinZhangBBL, 2025-07-29:
@layooesh Thank you for your feedback. We are looking in the issue. To get more information, could you reproduce the problem to offer log for us. you can refer to https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log