bambulab/BambuStudio #6778: The Calibration Result screen is empty after running Flow Dynamics calibration

FlewByYou, 2025-04-30:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 15.4.1

### Printer

Bambu Lab X1C (firmware 1.09.00.01)

### How to reproduce

1. Click on **Calibration**
2. Click on **Flow Dynamics** > **Auto Calibration**
3. Pick any color from AMS to run the calibration
4. When calibration is finished, click **Next** > 
5. Click **Manage Result**. The pop-up **Calibration Result** screen is empty

### Actual results

The Calibration Result screen is empty

[image]

### Expected results

The Results screen should show existing Flow Dynamics (k values) for existing filaments.

### Project file & Debug log uploads

[Bug.zip](https://github.com/user-attachments/files/19982243/Bug.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

alvesjc, 2025-05-02:
I'm having issues with calibration too.

I select the filament and start the process, but the printer doesn't load the filament to the toolhead and keeps doing the calibration like if it was loaded. In the end the test fails of course.

I've all the latest versions

FlewByYou, 2025-05-02:
> I'm having issues with calibration too.
> 
> I select the filament and start the process, but the printer doesn't load the filament to the toolhead and keeps doing the calibration like if it was loaded. In the end the test fails of course.
> 
> I've all the latest versions

Try resetting the printer to factory settings. This happened to me last night. All of sudden, Bambu Studio wasn't able to calibrate any filament and an error appeared on my X1C telling me to reset to factory settings. Did that and I am able to calibrate now (though the missing existing calibrations persist)

alvesjc, 2025-05-02:
> > I'm having issues with calibration too.
> > I select the filament and start the process, but the printer doesn't load the filament to the toolhead and keeps doing the calibration like if it was loaded. In the end the test fails of course.
> > I've all the latest versions
> 
> Try resetting the printer to factory settings. This happened to me last night. All of sudden, Bambu Studio wasn't able to calibrate any filament and an error appeared on my X1C telling me to reset to factory settings. Did that and I am able to calibrate now (though the missing existing calibrations persists)

I did it, because it installed new firmware, so I did factory reset and full calibration again.
Didn't help.

alvesjc, 2025-05-02:
This is really crazy, I now started the test with loaded filament already.
The printer did the bed calibration, and then unloaded the filament.
After unloading, I continue to see that the extruder yellow gear is running and off course nothing comes out.
Then it starts to print as if the filament would be loaded.....

How do I revert to previous fw ??

zhimin-zeng-bambulab, 2025-05-06:
@FlewByYou  Could you share your log files?

alvesjc, 2025-05-06:
@zhimin-zeng-bambulab , fyi, I upgraded mine again.

This time, I disconnected the AMS before starting the process, even being the old AMS and the instructions saying it was only needed for the new one.

This way, it did upgrade also the AMS's (I've 2).

After this update, the calibration I had done when downgraded disappeared again, and running calibration fails because the AMS doesn't pull the filament to be calibrated, so it's basically in the same state as it was in the 1st upgrade attempt.

Printing with AMS works fine so far.