bambulab/BambuStudio #6779: Slot #1 of AMS (on Prepare screen) shows incorrect filament (since last sync'ed)

FlewByYou, 2025-04-30:
### Bambu Studio Version

Bambu Studio 2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 15.4.1

### Printer

Bambu Lab X1C

### How to reproduce

1. Open Bambu Studio
2. Click on **Prepare**
3. Slot # 1 of the filament list shows incorrect filament (on the screenshot, it is showing **Bambu Lab PLA Basic** where it should have been **SUNLU PLA+**)
4. Hit **Synchonize** to sync the filaments. The correct filament is shown up (SUNLU PLA+)
5. Close Bambu Studio and re-open it. Slot #1 is showing wrong filament again

It is only on Slot #1. The other slots are working correctly.

[image]

### Actual results

Slot #1 of the AMS shows incorrect filament. Have to synchronize every time after opening Bambu Studio to show the correct filament. 

### Expected results

Slot #1 should show the correct filament (last synchronized) whenever Bambu Studio is opened.

[image]

### Project file & Debug log uploads

[Bug.zip](https://github.com/user-attachments/files/19982443/Bug.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2025-05-06:
Hello, we will try to reproduce and fix it

Haidiye00, 2025-05-06:
Hello, currently you can create a new cube and save this project. The consumables configuration will be recorded in the BambuStudio.exe file, and you won't need to synchronize it for the next time you open it