bambulab/BambuStudio #6781: Regression of 3mf bug

ondrovic, 2025-04-30:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Pro x64

### Additional system information

n/a

### Printer

Bambu Lab X1 C

### How to reproduce

1. Export a designed file from fusion
2. open in bambu studio

This is a regression of [this bug](https://github.com/bambulab/BambuStudio/issues/6256) - which wa resolved and now is back

### Actual results

[image]

### Expected results

There shouldn't be a warning for this, none of the older versions did this

### Project file & Debug log uploads

n/a 

### Checklist of files to include

- [x] Log file
- [x] Project file

Haidiye00, 2025-05-06:
Hello, can you provide this 3mf? Thank you

ondrovic, 2025-05-06:
> Hello, can you provide this 3mf? Thank you

It doesn't matter the .3mf if it's generated outside of Bambu Studio, in my case it was a file I build in Fusion Autodesk

Here are 4 test items all generated from Autodesk Fusion (360)

[Test 3mfs.zip](https://github.com/user-attachments/files/20061342/Test.3mfs.zip)

Haidiye00, 2025-05-06:
Hello, I have checked this 3mf and it is indeed not saved properly by our Bambustudio. So there will be this pop-up when importing.
Do you need to add an option to no longer remind, so that this pop-up window will not appear again next time?

ondrovic, 2025-05-06:
> Hello, I have checked this 3mf and it is indeed not saved properly by our Bambustudio. So there will be this pop-up when importing. Do you need to add an option to no longer remind, so that this pop-up window will not appear again next time?

Well of course they aren't saved by Bambu Studio, they are creating using Autodesk Fusion 360 and exported from there then imported into Bambu Studio to print.

Yes please give use the option to disable the pop up

closed 2025-07-18 by Haidiye00