bambulab/BambuStudio #6784: ?

kreamuhely, 2025-05-01:
### Bambu Studio Version

ghfuztdf

### Where is the application from?

Bambu Lab Official website, Bambu Lab github releases

### OS version

mac os

### Printer

P1s

### How to reproduce

1

### Actual results

Szar a rendszer és kibaszott mérges vagyok! Rohadtt drágán megvett nyomtatón nem lehet zökkenőmentesen filamentet váltani. Értelmetlen menürendszer, értelmetlen hibaüzenetek. Hagyják már a kibaszott AMS-üket! Nincs és ennyiért nem is akarok venni. A Anycubic sokkal olcsóbb és ugyan ezt tudja. Nincs AMS de oldják már meg rendesen a filament cserét!!!!!!
Fasssszom!

### Expected results

,jhfgihgf

### Project file & Debug log uploads

kjhfgizgf

### Checklist of files to include

- [ ] Log file
- [ ] Project file

JohnSmithCYM, 2025-05-06:
Sorry for the problem and can you offer more information to describe the problem, from the information you mentioned and you can not change the filament without AMS ?

BoBoYu-BamBu, 2025-05-22:
@kreamuhely Hello, may I ask if your problem has been solved?