bambulab/BambuStudio #6786: External Spool Auto Calibrations Broken

darkneon2002, 2025-05-01:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

win11

### Additional system information

i7 14700KF
64Gb DDR5
EVGA RTX2080Ti XC ULTRA

### Printer

X1C

### How to reproduce

Select printer in DEVICE tab
go to CALIBRATION tab
select any Calibration type
select AUTO-CALIBRATION
select EXTERNAL SPOOL
press CALIBRATE

### Actual results

screen for calibration progress will show
calibration will not start
no error will show
no time-out will occur
progress will remain at N/A

### Expected results

Calibration should start and finish

### Project file & Debug log uploads

[Untitled.zip](https://github.com/user-attachments/files/20000139/Untitled.zip)

[log_iotc.zip](https://github.com/user-attachments/files/20000151/log_iotc.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

Skarules, 2025-05-09:
About the same problem Got a Error with Bambu Studio 2.0.3.54, invalid tray_id or use_ams flag when trying with external spool.
Tried to reinstall version  2.02.57 still having the error. Guess its come from new firware of the printer ? 
Printer is X1C V 1.09.00.01

zhimin-zeng-bambulab, 2025-05-10:
same as https://github.com/bambulab/BambuStudio/issues/6776

vanowm, 2025-06-01:
@zhimin-zeng-bambulab do you have authority to close issues? This is clearly a duplicate...no wonder there are 4k opened issues here, nobody seems to bother organize them...