bambulab/BambuStudio #6789: Filament and Process Change to Default When Switching Back and Forth Between Printers

jagc615, 2025-05-01:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS Ventura 13.0.1

### Printer

Bambu P1P and A1

### How to reproduce

Open file with P1P. All settings loaded as saved.
Switch printer to A1.
Switch printer back to P1P. 

### Actual results

After changing to a different printer, when I go back to the original printer all filament goes back to Bambu PLA Basic instead of my previously set filament. Process goes back to Bambu Standard P1P.

[image]
[image]
[image]

### Expected results

Bambu Slicer should save my previous settings so I can switch between printers and have my settings be there when I switch back.

### Project file & Debug log uploads

[For Bambu.3mf.zip](https://github.com/user-attachments/files/20002898/For.Bambu.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

ThorsReign, 2025-05-01:
This was behavior in bambu studio prior to 2.0 and should continue today, I second that this is a bug.  Please fix.

bambu123, 2025-05-08:
A1/P1P are not compatible, the saved custom presets can only be compatible with the machine when saved

ThorsReign, 2025-05-08:
The issue is that the P1 preset you were on when you switch to the A1 isn't selected when you go back to the P1.  In Bambu 1.x versions the presets are maintained as you switch and return.