bambulab/BambuStudio #6791: Cannot access fan speeds

Harley335, 2025-05-02:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

intel core i7-12700k 
32gb RAM
RTX 3070TI

### Printer

Bambu Lab X1C

### How to reproduce

Click "Device" on top bar
click "fan"

### Actual results

When clicking on the fan controls there are no actual controls for the fan speeds.

[image]

### Expected results

After updating my X1C to the new firmware, I cannot control my fan speeds from Bambu Studio.

### Project file & Debug log uploads

[Fan Speed.zip](https://github.com/user-attachments/files/20008786/Fan.Speed.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

Duudeeee, 2025-05-02:
and on handy app the control fans its crazy

ldigirolamo, 2025-05-04:
same on my setup

kle7rx, 2025-05-05:
Same problem. X1C printer.

[image]

shuizhiguo, 2025-05-07:
**Please upload log through Handy APP, and give us your ticket number**
***
***Upload log files through Bambu Handy***
     Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
     When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
     To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation.
     [image]

Harley335, 2025-05-07:
> **Please upload log through Handy APP, and give us your ticket number**
>
> _**Upload log files through Bambu Handy**_ Log files from the Bambu printer can be uploaded via Bambu Handy over the network, [...]

Here is my ticket number - US250507069001

Please let me know if any additional information is required.

Thank you.

Aipoc76, 2025-05-26:
Same issue.

closed 2025-05-29 by BoBoYu-BamBu

BoBoYu-BamBu, 2025-06-04:
Sorry, this issue is related to the printer. Please follow the above instructions to create a service order. The subsequent follow-up will be handled by the colleagues from the firmware department. Furthermore, the owner of this order has already created an after-sales report, so this order has been closed.

> **请通过 Handy APP 上传日志，并提供您的票号**
> 
> _**通过 Bambu Handy 上传日志文件。Bambu**_ 打印机的日志文件可以通过 Bambu Handy 通过网络上传，在最佳网络条件下，上传速度约为 150 kbps。 上传日志时，您可以选择特定的时间段，从而减少传输的日志量并缩短上传时间。 要上传文件，请前往**“我的”** →** “支持工单”** →“选择要上传日志的工单”，然后继续操作。 [image]

Duudeeee, 2025-06-04:
> hilarious, they close the other ticket as a dupe of this, and they close this as a dupe of the other ticket
> 
> Fan speed no workieda. aka You can not adjust the fan speeds via PC Software.

yeah u right, i already open ticket long time ago and never got replay.