bambulab/BambuStudio #6794: bambustudio crashing!!!!!

tyomabondar, 2025-05-02:
### Bambu Studio Version

01.00.03.54

### Where is the application from?

Bambu Lab Official website

### OS version

OS:macOS 11 (big sur)

### Printer

what

### How to reproduce

it just freezes
on my case its to open bambustudio and wait some time

### Actual results

i see a white box

### Expected results

idk

### Project file & Debug log uploads

my 3mf file (https://github.com/user-attachments/files/20008360/icy.3mf.zip)

log (https://github.com/user-attachments/files/20011631/debug_network_Fri_May_02_07_24_35.log.enc.zip)

ad .diag file

### Checklist of files to include

- [x] Log file
- [x] Project file

BoBoYu-BamBu, 2025-05-06:
@tyomabondar  Thank you for your feedback. Could you please provide the DMP file? Were there any other operations performed before the crash?

davidjdixon, 2025-05-06:
@BoBoYu-BamBu could you please stop asking macOS users for a DMP file. That is a Windows-specific file and the macOS version stores log data differently, and not to a specific file. It would be helpful if you could provide platform-specific requests for further information if it is needed.

Haidiye00, 2025-05-12:
Hello, can you take a screenshot

tyomabondar, 2025-05-26:
here is the screenshot:[](https://drive.google.com/file/d/1jAaY2Wkdy0XiRaaNJyw8OpA2snDhjpoO/view?usp=sharing)

Haidiye00, 2025-05-27:
Hello, sorry, I am unable to open the file you uploaded.
Hello, can you download ScreenToGif software and record a GIF of the operation video and upload it? Thank you
[Release ScreenToGif 2.41.1 · NickeManarin/ScreenToGif](https://github.com/NickeManarin/ScreenToGif/releases/tag/2.41.1)

tyomabondar, 2025-06-01:
fixed the issue by deleting user data and reinstalling bambu studio.

closed 2025-06-01 by tyomabondar