bambulab/BambuStudio #6798: Problem with adjusting flush volumes on Mac OS

Zach-coder, 2025-05-02:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

macOS Sequoia 15.4.1 (24E263)

### Printer

Bambu Lab P1S

### How to reproduce

Try to adjust Flushing volumes.

### Actual results

Pulls up a very small BBS window that stays completely blank, there is no option to minimize this window.

### Expected results

I should be able to adjust the flushing volumes and multiplier for the filaments in my AMS. 

### Project file & Debug log uploads

The project file used does not matter, and this issue does not cause BBS to crash. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-BamBu, 2025-05-06:
@Zach-coder  Sorry, could you please provide a screenshot or a video recording to describe your issue?

XunZhangBambu, 2025-05-10:
@Zach-coder  Same as #6282. When installing, you need to drag Bambu Studio into the application folder and do not change the name of the application folder.