bambulab/BambuStudio #6801: Status lüfterX1C

racerman76, 2025-05-03:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

Cpu: AMDRyzen 9 5950x 
Ram:32 GB

### Printer

Bambu Lab X1C

### How to reproduce

Kein staus der Lüfter in der software 

### Actual results

Seit Update 01.09.00.01 auf X1C wird keine Zustand der Lüfter in der Statusübersicht in der Software mehr angezeigt .

### Expected results

Anzeige Staus Lüfterzustand

### Project file & Debug log uploads

[debug_Sun_Apr_27_10_29_21_4456.log.zip](https://github.com/user-attachments/files/20022866/debug_Sun_Apr_27_10_29_21_4456.log.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

racerman76, 2025-05-04:
Jetzt kommt nur noch das Fenster.

[image]

sristow646, 2025-05-05:
I have the same Issue on my MAC.
mcOS: Sequoia 15.4.1

racerman76, 2025-05-07:
Ticket No: US250507105001