bambulab/BambuStudio #6803: A1-mini without ASM - Filament preset is not listed - can't print

xxxAleksandrxxx, 2025-05-03:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Built from source by yourself or others

### OS version

macOS 14.6.1 build 23G93

### Additional system information

MacBook Pro M1

### Printer

A1-mini

### How to reproduce

select Bambu PETG-CF filament
slice
print

### Actual results

Can't send gcode sliced with PETG-CF or other filaments (except PLA). BambuStudio asks to select filament but doesn't show it.

[image]
[image]

but if it was PLA, it is listed

[image]

I tried the following steps didn't help
- Reinstall BambuStudio
- delete BambuStudio, delete configuration from (.../Application Support/BambuStudio) and installing it agan

### Expected results

can't select proper filament and "send" button is inactive

### Project file & Debug log uploads

[Cylinder_D7.8.3mf.zip](https://github.com/user-attachments/files/20023889/Cylinder_D7.8.3mf.zip)

[log.zip](https://github.com/user-attachments/files/20023893/log.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-BamBu, 2025-05-06:
@xxxAleksandrxxx  Sorry, we are currently addressing this issue. You can first manually set the extrusion consumables to PETG-CF in the device page, then send the print job. It should be possible to print with the extrusion materials now.

XinZhangBBL, 2025-06-26:
Halo, we have solved the problem at latest release. please check https://github.com/bambulab/BambuStudio/tags