bambulab/BambuStudio #6805: Tiles to large to be shown on screen

gruendling, 2025-05-03:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

I have a laptop with 2880 x 1800 resolution.

In windows the recommendet setting 

**200%**

[image]

[image]

**175%**

[image]

[image]

### Printer

Bambu Lab X1C

### How to reproduce

see screenshots

### Actual results

see screenshots

### Expected results

see screenshots

### Project file & Debug log uploads

see screenshots

### Checklist of files to include

- [ ] Log file
- [ ] Project file

XinZhangBBL, 2025-05-06:
@gruendling  Halo, the font scale is too large. You can use the scroll bar to view the items
[image]

BoBoYu-BamBu, 2025-05-22:
@gruendling  Hello, may I ask if your problem has been solved?

gruendling, 2025-05-22:
HiThanks for your message. It’s not a solution but I switched back to small font size. Besides is it true that firmware was blocked for third party updates? This is really a regrettable move. RegardsTillDr. Till GründlingOffenbacher Str. 1768305 Mannheim0170/6000368Am 22.05.25 um 10:25 schrieb BoBoYu-BamBu
                
            
            
                
                    Von: "BoBoYu-BamBu" ***@***.***>Datum: 22. Mai 2025An: "bambulab/BambuStudio" ***@***.***>Cc: "Mention" ***@***.***>,"gruendling" ***@***.***>Betreff: Re: [bambulab/BambuStudio] Tiles too large to be shown on screen at 200% scaling factor (Issue #6805)
                
                
BoBoYu-BamBu left a comment (bambulab/BambuStudio#6805)

@gruendling  Hello, may I ask if your problem has been solved?