bambulab/BambuStudio #6807: updated software and now it crashes

lslaws8, 2025-05-03:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Mac OS sequoia 15.4.1 

### Printer

Bambu lab X1 series

### How to reproduce

[image]

### Actual results

[image]

### Expected results

the print files from Makers World will download to bambu studio

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/20024305/log.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

Haidiye00, 2025-05-06:
Hello, can you download ScreenToGif software and record a GIF of the operation video and upload it? Thank you
[Release ScreenToGif 2.41.1 · NickeManarin/ScreenToGif](https://github.com/NickeManarin/ScreenToGif/releases/tag/2.41.1)

kazthehack, 2025-06-09:
still an issue on my end, with the latest version of the app. Its even worse tbh, i cant go in prepping one model without a crash in the following steps

- setting up custom filament configuration
- setting up filament information on the device tab 
- calibrating the custom filamets
- trying to print with the filaments (after slicing)

Haidiye00, 2025-10-15:
https://github.com/bambulab/BambuStudio/releases/tag/v02.03.00.70
Hello, please update the software version.

Haidiye00, 2025-11-19:
https://github.com/bambulab/BambuStudio/releases/tag/v02.04.00.70
Hello, the new version has been released, please update.
If there are any further questions, let's continue the discussion. Thank you