bambulab/BambuStudio #6808: Auto Flow Dynamics Calibration doesn't work with 0.6mm nozzle

Piropr, 2025-05-03:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Win 10 64 bit

### Printer

X1C

### How to reproduce

1. Set nozzle diameter as 0.6mm in Device tab and printer settings.
2. Go to Calibration>Flow Dynamics Calibration
3. Select Auto-Calibration
4. Sync AMS and nozzle information

[image]

6. Confirm Nozzle Diameter to 0.6mm

[image]

8. Select AMS slots
9. Run calibration

### Actual results

Generic error code: "Selected diameter and machine diameter do not match."

[image]

### Expected results

Auto Calibration will run.

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [x] Log file
- [x] Project file

shiftylilbastrd, 2025-05-07:
This isn't limited the the 0.6mm nozzle, I get the same error with the 0.4mm.

zhimin-zeng-bambulab, 2025-05-10:
Thank you for reporting this issue. We will investigate the problem. Which firmware version are you using?

zhimin-zeng-bambulab, 2025-05-10:
And could you share your log files?

Piropr, 2025-05-10:
Hi, I'm using firmware version 01.09.00.01. Uploaded the log as well.

[debug_Sat_May_10_09_43_44_25916.log.zip](https://github.com/user-attachments/files/20140031/debug_Sat_May_10_09_43_44_25916.log.zip)

shiftylilbastrd, 2025-05-10:
[image]
I think this is the right log
[debug_Fri_May_09_17_00_11_10092.log](https://github.com/user-attachments/files/20144305/debug_Fri_May_09_17_00_11_10092.log)

koko1111, 2025-05-11:
I am also having this issue. Confirmed multiple times that all nozzles are set the same. Restarted everything. Deleted and added back printer. For me I was going from a 6 to a 4 nozzle. If I change back to 6, it works.

s0ds0ds0d, 2025-05-14:
Same issue for 0.8mm on X1C (FW: 01.09.00.01)

Alan-Somebody, 2025-05-17:
I am having the same issue.  Trying to Auto Calibrate on my X1C using a 0.6mm nozzle, and it comes up with the error.  I have confirmed that the X1C on it's built-in touch-screen shows that it has a hardened steel 0.6mm nozzle.  This is also the same nozzle size selected in the auto calibration screen.

Bambu Studio version: 2.0.3.54

[image]

t6d, 2025-05-18:
I'm running into the same issue with one of my two X1Cs. I already tried

* downgrading the firmware,
* re-upgrading to the latest firmware version,
* factory reset of the machine,
* deleting `~/Library/Application Support/BambuStudio`

None of the above has resolved my issue. I can only calibrate with a 0.4mm nozzle. The frustrating part, my second X1C gives me zero issues.

picaron, 2025-05-19:
I have same error trying to run calibration with 0.6mm hardened steel nozzle

running latest firmware as well

changbaishen, 2025-05-20:
**Please upload log through Handy APP, and give us your ticket number**
***
***Upload log files through Bambu Handy***
     Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
     When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
     To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation.
     [image]

ageitgey, 2025-05-20:
> Believing the issue was fixed I upgraded back to the latest firmware and tried calibration again. No longer getting the nozzle error but now the AMS isn't loading filament. Have all four slots selected for calibration and none of them load.

I have the same problem where it won't feed filament from the AMS before running calibration. Seems to be the issue here: #6880

steckdev, 2025-05-21:
I have the same issue. Looks like I need to downgrade again. I also had this issue of where the color would not swap for the first couple colors. But was also experiencing other issues. But this was really confusing working through a jammed nozzle... Looking forward to the fix.
I tried swapping settings mutliple times, syncing multiple times, and the ui still won't honor that I have .4 on the machine, and .4 in the bambu studio.