bambulab/BambuStudio #6809: Cloudfalre Verify Human Infinite Loop

arejaybee, 2025-05-03:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Built from source by yourself or others

### OS version

PopOS

### Printer

Bambu Lab P1S

### How to reproduce

- Launch Bambu Labs
- Click Online Models, Click MakerLab, or search anything from Home page
- The makerworld.com page is displayed asking the user to verify they are human
- User clicks the Verify you are human checkbox

### Actual results

- The spinner loads for a while
- Then the box refreshes to verify you are human
- No matter how many times the user clicks this, they are never verified
- The application is genuinely unusable as this locks the user out of all features

https://github.com/user-attachments/assets/d8561b4d-3719-4b3d-8b1d-80c9519d666c

### Expected results

The user is verified and navigates to the page

### Project file & Debug log uploads

[log_iotc.txt](https://github.com/user-attachments/files/20024724/log_iotc.txt)

Note:
This issue persists after:
- system reset
- application uninstall/reinstall
- application version update

### Checklist of files to include

- [x] Log file
- [ ] Project file

arejaybee, 2025-05-04:
I am unsure if this helps, but there is an issue with the storage calls. 

[image]

LynxLo, 2025-05-04:
I gave the same printer(P1S) and the same OS(Pop!_OS 22.04). Let me know if I can help put in any way.

cpetry, 2025-05-04:
Got the same problem.
Zorin OS, Bambu A1

nmendiboure, 2025-05-04:
Got the exact same issue. Linux Mint 22.1 (kernel 6.11) + bambustudio 2.0.3.54 flatpak version.

Moreover I can't load a model from the online models, it says "404 this model does not exist or has been deleted, please refresh the page and try again". I don't know if both issue might be linked. 

I tried different networks, changing the language, changing the region. Nothing worked yet.

nmendiboure, 2025-05-04:
EDIT : it seems to work with the ubuntu 24.04 appimage. So maybe it's relative to the flatpak version.

cyclotron3k, 2025-05-06:
I have this problem too. v2.0.3.54 AppImage on Ubuntu 24.04.2 LTS

MackBambu, 2025-05-08:
Is it normal to access the MakerWord website through a browser?

arejaybee, 2025-05-08:
> Is it normal to access the MakerWord website through a browser?

I have never tried to access it through a browser. This issue is occurring within the BambuStudio client.

MackBambu, 2025-05-08:
> > Is it normal to access the MakerWord website through a browser?
> 
> I have never tried to access it through a browser. This issue is occurring within the BambuStudio client.

Could you try accessing makerworld.com directly through a browser and logging into your account? I’d like to rule out whether it’s a cloud service issue related to your region.

arejaybee, 2025-05-08:
> Could you try accessing makerworld.com directly through a browser and logging into your account? I’d like to rule out whether it’s a cloud service issue related to your region.

Logged in and confirmed the browser works correctly and that the issue is still present in the application.

MackBambu, 2025-05-08:
@arejaybee Is the problem with Flathub, AppImage, or both?

MackBambu, 2025-05-08:
@hadess When running libwebkit under Flathub, it seems to trigger Cloudflare's JavaScript Challenge. Do you have any experience with this?

hadess, 2025-05-08:
Looping Cloudflare captchas is a common WebKit problem, it's not related to Flatpak, or WebKitGTK:
https://bugs.webkit.org/show_bug.cgi?id=265874

I don't know if https://upload-file.bambulab.com/ and https://upgrade-file.bambulab.com/ not working is related (those also don't work in a browser), but it throws 404 error in a JSON file which could definitely be one of the causes of this problem.

Could you please get those 2 URLs fixed before we investigate further?

MackBambu, 2025-05-08:
@hadess

> I don't know if https://upload-file.bambulab.com/ and https://upgrade-file.bambulab.com/ not working is related (those also don't work in a browser), but it throws 404 error in a JSON file which could definitely be one of the causes of this problem.

upload-file.bambulab.com and upgrade-file.bambulab.com are both deprecated addresses. The "storage test" feature has also been deprecated, but we forgot to remove it.

The cloud services team has found a large number of interception logs, all triggered by the JavaScript challenge. It appears that the AppImage version is working fine, but the Flathub version is being blocked.

https://medium.com/@datajournal/cloudflare-js-challenge-how-to-solve-83c2f02b92e1

m-m-adams, 2025-05-09:
Also having this issue on fedora 41 with the latest flatpak from flathub

MrSiO, 2025-05-09:
Any of you using "AdGuard" ?

obc999, 2025-05-10:
Fedora 42 here also having the issue. Though it was a fedora issue and have tried it in pop.os, linux mint, and ubuntu. an absolute newb when it comes to these things so im not quiet sure how to fix.

jamesw, 2025-05-11:
I can confirm that this is a Bambu Studio issue, I can also confirm that this seems to be related to Linux, running Linux mint I get this error, running Windows 10 in the same machine same account I do not see this issue, I am able to browse using various browsers in Linux mint and download and send to banbu studio with no issues, this is definitely a studio/Linux issue

MackBambu, 2025-05-13:
We've adjusted the Cloudflare policy, is MakerWorld loading properly now?

hadess, 2025-05-13:
> We've adjusted the Cloudflare policy, is MakerWorld loading properly now?

Works for me.

[image]

MackBambu, 2025-05-14:
> > We've adjusted the Cloudflare policy, is MakerWorld loading properly now?
> 
> Works for me. Any chance you could tell us what you changed to make this work?
> 

There wasn't a better solution — we just added a whitelist and excluded the Flatpak version of the browser.

LynxLo, 2025-05-17:
Works now! Thanks a lot guys for fixing it.

closed 2025-05-20 by MackBambu

Oldnumber9, 2025-07-03:
Bambu noob here.  Experiencing this problem, cannot access makerworld via Bambu studio.

Verified I can log into makerworld via firefox.

Using:

NAME="Nobara Linux"
VERSION="42 (KDE Plasma Desktop Edition)"
ID=nobara
ID_LIKE="rhel centos fedora"

Bambu Studio          2.1.1 Public Release     stable      flathub      system

Browser Firefox

arejaybee, 2025-07-05:
As stated above this issue is happening agian. I'll log a new ticket and link to this.