bambulab/BambuStudio #6810: Bambu studio 2.0.3.54 hangs while computing infill pattern

mdebarwepion, 2025-05-03:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu A1

### How to reproduce

Bambu Slicer (2/0/3/54) hangs when slicing this model : https://makerworld.com/en/models/1243666-odin-kniend-mit-hugin-und-munin-statue?from=search#profileId-1264633. It hangs showing 35% progress. This happens on two different Windows 11 machines, one of them with much more memory, faster cpu, different nvidia graphics card. I tried various combinations of infill patterns and percentage to no avail. Orca slicer 2.3  does the slicing without any problem.

### Actual results

Slicing hangs

### Expected results

Slicing completes

### Project file & Debug log uploads

[odin+3mf.zip](https://github.com/user-attachments/files/20025180/odin%2B3mf.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

ThorsReign, 2025-05-06:
I recommend downloading the STL/CAD files instead of the print profile and trying again - I just did that and sliced no problem.

mdebarwepion, 2025-05-06:
Thank you for this suggestion. I tried it and it does indeed work. I had
never yet encountered this problem and I did not think of trying the stl
file rather than the 3MF.

I printed the version sliced by Orca Slicer, using normal arborescent
supports. They were fiddly to remove, and I had one or two minor problems
of breakages when removing them. They will be fixed.

I admire the model very much

Michel

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Sans
virus.www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

QingZhangBambu, 2025-07-21:
[image]
it worls fine on the latest version, could you try it?