bambulab/BambuStudio #6811: Display problem with Color Painting Tool

Tony-KY, 2025-05-04:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

OS: Windows 11 Pro

### Additional system information

CPU: 13th Gen Intel(R) Core(TM) i9-13900KF   3.00 GHz
Memory: 64.0 GB
Display: NVIDA GeForce RTX 4070 Ti 12 GB

### Printer

Bambu Lab H2D

### How to reproduce

1. Goto Prepare Tab
2. Select Color Painting Tool
3. Use Painting Tool on 3d object, sometimes the tool will show the color change sometimes not
4. Goto Preview Tab, This tab will show the color change even though it is not shown in the Prepare Tab

[image]
[image]

### Actual results

When using the Color Painting Tool to paint your object after selecting the object it does not show the color change. This does not happen all the time. Going to the Preview Tab does show the color change even though it is not shown in the Prepare Tab. The two images I sent show this issue, no color change in Prepare and with color change in Preview Tab.

### Expected results

When using the Painting Tool in Prepare Tab the color should change on the object selected with the tool.

### Project file & Debug log uploads

Project File: 

[New Bird House v4.zip](https://github.com/user-attachments/files/20025526/New.Bird.House.v4.zip)

Log File: 

[debug_Sat_May_03_20_50_32_20916.log.zip](https://github.com/user-attachments/files/20025528/debug_Sat_May_03_20_50_32_20916.log.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

Haidiye00, 2025-05-06:
Hello, can you download ScreenToGif software and record a GIF of the operation video and upload it? Thank you
[Release ScreenToGif 2.41.1 · NickeManarin/ScreenToGif](https://github.com/NickeManarin/ScreenToGif/releases/tag/2.41.1)

Tony-KY, 2025-05-06:
Attached is the Gif file you requested.

[Painting Display Issue in Prepare Tab.zip](https://github.com/user-attachments/files/20065837/Painting.Display.Issue.in.Prepare.Tab.zip)

Tony-KY, 2025-05-06:
I think I resolved this issue by applying fix model to this project.

Haidiye00, 2025-05-07:
Hello, I have reviewed your GIF and confirmed that the coloring is normal after repairing it. The number of triangles in the repaired patch has decreased, and it seems that there are some invisible issues with the original triangle structure~

closed 2025-08-02 by Haidiye00