bambulab/BambuStudio #6812: Please replace the nozzle message when printing PETG-CF

JeffGoldner, 2025-05-04:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS 15.5 Beta 2

### Printer

Bambu Lab P1S

### How to reproduce

Load BL PETG-CF spool in AMS
In device tab, printer parts, the nozzle type is set to "Hardened Steel", 0.4 - which is true, I replaced the stainless nozzle months ago and set this.
When you do a Print Plate, the warning is displayed "The nozzle hardness required by the filament is harder than the default nozzle hardness of the printer, etc."

Why not read the actual nozzle from the settings? There is nothing offered by BL that is "harder" than the hardened steel nozzle.

### Actual results

Click Confirm and it prints fine

### Expected results

No warning/error should be displayed.

### Project file & Debug log uploads

This was just using the built-in primitives in B/S - a cube and benchy

### Checklist of files to include

- [ ] Log file
- [ ] Project file

XinZhangBBL, 2025-05-07:
@JeffGoldner  Thank you for your feedback. We are looking in the issue. To get more information, could you reproduce the problem to offer log and video/screenshots for us. you can refer tohttps://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log

XinZhangBBL, 2025-06-17:
Please try with latest bambu release。You can reopen the problem if the same issue occurs

ts8801, 2025-09-19:
@XinZhangBambu I am still getting this pop up in 2.2.2.56, Printing PETG-CF without AMS with a .4 hardened steel nozzle set .

slowthisbirddown, 2025-10-31:
Still literally an issue