bambulab/BambuStudio #6815: Latest Update Flaw

Tmprmntl13, 2025-05-04:
### Bambu Studio Version

Bambu_Studio_win-v02.00.03.54

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Home (23H2)

### Additional system information

I just finished updating myX1 Carbon machine where it claims to fix AMS bugs. When I went to run a project I noticed that my four filaments are shown but listed in an incorrect order. After re-sync'ing several times, the #4 filament is shown in the #1 position in the AMS. When I send a project from Bambu Studio, it shows the filament in the correct (#4) position (see attached screenshot). Please review and correct, it is very confusing.
Respectfully,
D Monroe

[image]

### Printer

Bambu Lab X1-Carbon

### How to reproduce

I have shut down both X1 Carbon machine and laptop, restarted everything in different sequences, and re-sync'd Bambu Studio app to machine...several times. Nothing changes this glitch.

### Actual results

Please help me Obi-Wan, you're my only hope.

### Expected results

When re-sync'ing filament, it should show what is in the AMS, in correct order. In my case, I have black in #1, white in #2, light blue in #3, and yellow in #4. If you look at the picture, it is showing the yellow as #1, pushing the other filaments to the next number. But when sending (again, seen in the attached picture) it is correctly identifying yellow as being in slot #4.

### Project file & Debug log uploads

[log_iotc.txt](https://github.com/user-attachments/files/20028126/log_iotc.txt)

[Soap Dish Clip-on.zip](https://github.com/user-attachments/files/20028141/Soap.Dish.Clip-on.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

JeffGoldner, 2025-05-04:
When you resynch, select Overwriting instead of Mapping. That should be the default IMO.

Tmprmntl13, 2025-05-04:
That worked like a champ. Thank you so much. 
Respectfully,D Monroe

BoBoYu-BamBu, 2025-05-06:
@Tmprmntl13  May I ask if your problem has been solved? Can we close this issue now?

closed 2025-05-09 by BoBoYu-BamBu