bambulab/BambuStudio #6817: Issue with cloning object

irDragon, 2025-05-04:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 ver 10.0.26100

### Additional system information

Processor	AMD Ryzen 9 5900 12-Core Processor, 3001 Mhz, 12 Core(s), 24 Logical Processor(s)
Installed Physical Memory (RAM)	16.0 GB

### Printer

Bambu Labs P1S

### How to reproduce

1: Select object
2: Right click (for sub menu)
3: Select clone
4: Select "Scale" on top menu bar
5: Compare scale of both the original and the clone

### Actual results

[image]
[image]

### Expected results

The cloned object should be the same dimensions as the original.

### Project file & Debug log uploads

[Cloning issues.zip](https://github.com/user-attachments/files/20028929/Cloning.issues.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2025-05-06:
Hello, can you download ScreenToGif software and record a GIF of the operation video and upload it? Thank you
[Release ScreenToGif 2.41.1 · NickeManarin/ScreenToGif](https://github.com/NickeManarin/ScreenToGif/releases/tag/2.41.1)

irDragon, 2025-05-09:
Currently, I can not repeat the issue.  When it occurs again I will attach
a video of the operations and send it to you.

Thanks,
Steve

Haidiye00, 2025-08-04:
https://github.com/bambulab/BambuStudio/releases
Hello, I was also unable to reproduce this issue
Please use the latest version. If there are still any issues, we will continue to discuss. Thank you

closed 2025-08-04 by Haidiye00