bambulab/BambuStudio #6819: [Bug] Bambu Handy 3.1.0 Not Displaying Fan Outputs Correctly on X1C with Firmware 01.09.00.01

StinkyLinky8, 2025-05-05:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 Pro 19045.5737

### Additional system information

ThinkPad Yoga X1

### Printer

Bambu Labs X1C

### How to reproduce

Steps to reproduce the behavior:
1. Update X1C to firmware 01.09.00.01
2. Install Bambu Handy app version 3.1.0 on iOS 18.4.1
3. Start a print
4. View live fan outputs in the Bambu Handy app
5. Observe incorrect or static fan values

### Actual results

Fan readings are incorrect, frozen, or completely missing, even though the fans are visibly running and reporting correctly on the printer touchscreen and in Bambu Studio.

### Expected results

The Handy app should display real-time, accurate fan output values that match actual fan activity on the printer.

### Project file & Debug log uploads

[mousetrap2.zip](https://github.com/user-attachments/files/20030687/mousetrap2.zip)

[debug_Mon_Apr_14_13_31_33_14184.log.zip](https://github.com/user-attachments/files/20030688/debug_Mon_Apr_14_13_31_33_14184.log.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

shuizhiguo, 2025-05-07:
**Please upload log through Handy APP, and give us your ticket number**
***
***Upload log files through Bambu Handy***
     Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
     When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
     To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation.
     [image]

niklerus, 2025-05-12:
Noticed almost the same thing on my P1S, the part fan reports 0 even though I visually confirmed it was running (paused and removed extruder cover and checked that the parts fan was running).
Using the android version of Bambu Handy (v3.1.0)
Firmware in P1S, 01.08.00.00

MilanBlairon, 2025-05-30:
Same thing here

BoBoYu-BamBu, 2025-06-03:
> **Please upload log through Handy APP, and give us your ticket number**
>
> _**Upload log files through Bambu Handy**_ Log files from the Bambu printer can be uploaded via Bambu Handy over the network, [...]

Sorry，This issue will be followed up in this after-sales order as it is a problem with the printer. If no after-sales order has been created, please follow the above guidelines to create one.