bambulab/BambuStudio #6820: Adding filament change using MacBook Pro trackpad

cooper7012, 2025-05-05:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 15.4.1

### Printer

X1C & P1S

### How to reproduce

1. Must be using a macBook trackpad
2. click on slice plate
3. move the layer slider to the desired layer to add a filament change
4. try to right click on the + to add a filament change.

### Actual results

With the track pad I'm unable to right click on the + to add a filament change. The only workaround I can find is to connect a mouse to my laptop and right click with a mouse.

### Expected results

It would be nice if I could add a filament change using the MacBook trackpad

### Project file & Debug log uploads

I have this issue with all projects

### Checklist of files to include

- [ ] Log file
- [ ] Project file

cooper7012, 2025-05-05:
Although right-click doesn't work when trying to add a filament change using a trackpad, I just discovered a workaround — Control + Click does the trick!

Haidiye00, 2025-05-12:
Hello, could you please take a screenshot of the consumables replacement you mentioned?

gnorbsl, 2026-10-02:
This is still an issue with macs, right click also moves the layer slider, clicking the green plus does nothing