bambulab/BambuStudio #6821: 用输入数值的方式调整多个子零件坐标产生的结果位置错误

moe001, 2025-05-05:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

无关

### Additional system information

无关

### Printer

p1s

### How to reproduce

1. 载入工程文件
2. 在左侧对象窗口按住ctrl，同时选中两个黑色的立方体
3. 在上侧工具栏选择移动
4. 切换至世界坐标，并在z栏输入1.5并按回车

### Actual results

可以看到两个黑色方块并没有被移动到高度1.5mm附近，甚至还产生了反向移动。

### Expected results

正确将两个模型的共有高度中点放置在z1.5mm处

### Project file & Debug log uploads

[立方体.zip](https://github.com/user-attachments/files/20032883/default.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

Haidiye00, 2025-05-06:
Hello, it's like this, the xyz of multiple objects may not be equal, so we have set it uniformly to 0,0,0. You can modify it for a single object. Of course, when multiple objects are modified, the coordinates of the objects in 3D space have already changed, but there is no change on the xyz display

moe001, 2025-05-06:
但这两个黑色立方体的z是相同的，这种情况下应该支持同时选中它们并直接设置它们的z方向位置。

我理解当前你说的0,0,0逻辑，但在相同用户界面中，对于单个对象和多个对象的行为不一致，很令人迷惑。选中单个对象时输入1.5，就可以把模型设置到z=1.5，但选中两个时，输入1.5，模型被设置到了世界坐标z=6.5，还有比这更离谱的事情吗？这就是bug。

哪怕禁止同时修改多个模型的高度也比现在更加直白，当然最好的办法，我认为是以多个模型的外包络中点作为设置坐标的中点。

我经常需要把svg生成的大量小区块(n>30)移出原平面，进行涂画或者切分后再移动回去，依次对单个模型设置坐标实在是太繁琐，现在只能依赖于这个基于0,0,0的算法去测试着移动，因为很容易算错。

Haidiye00, 2025-05-07:
您好，我们内部讨论和评估一下

moe001, 2025-05-14:
有后续吗？似乎每当提到内部讨论的时候，就再也没有音讯了。

JohnSmithCYM, 2025-05-15:
感谢您的反馈。我们已对问题进行了讨论并有初步方案，在开发和测试工作结束后随未来版本发布，再次感谢您的反馈。