bambulab/BambuStudio #6823: Mysterious message box

carstenpoulsen, 2025-05-05:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

win11

### Printer

X1

### How to reproduce

When loading a 3MF file I get the message:
"The 3mf is not from Bambu Lab, load the geometry only"

### Actual results

What is the options?
I can only choose "OK" and the 3mf file loads fine.
Why do I get this completely useless message?
No matter what, the next is that the 3mf i loaded no matter if it sent to me from Bambu Labs or I create it myself.
What I the reason that I have to acknowledge that it is not from Bambu Labs???

Please remove this completely meaningless message.

### Expected results

No meaningless messages that I have to acknowledge

### Project file & Debug log uploads

Load a 3mf .... probably not one from Bambu Labs

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2025-05-06:
Hello, can you provide the 3mf that caused the pop-up window? Thank you

carstenpoulsen, 2025-05-06:
Sure, I provided it with the case.

Haidiye00, 2025-05-06:
Hello, do you mean we won't remind this pop-up next time? Can you tell me how this 3mf is generated
[image]
The following is normal
[image]

closed 2026-09-08 by carstenpoulsen