bambulab/BambuStudio #6825: Printer Advanced Settings not available

benisbell, 2025-05-05:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS Sequoia 15.4.1

### Additional system information

CPU: Apple M1
Memory: 16gb

### Printer

P1P upgraded to P1S

### How to reproduce

1. Selected Printer
2. Click the Edit Settings button
3. in the settings dialog, click on the Advanced Switch

### Actual results

Clicking on the Advanced switch does NOTHING. The switch does not respond at all. Advanced settings are not displayed and are not reachable. I can access the Advacned settings in Orca Slicer without any issue, but not in Bambu Labs Studio.

### Expected results

The Advanced switch should visually change to show it is on and the Advanced settings controls should be displayed and accessable.

### Project file & Debug log uploads

No project file is needed. This is in the Printer settings. No log file can be created as there is nothing to log. The application feature is not responding.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-BamBu, 2025-05-06:
@benisbell  Sorry, could you please provide a screenshot or a video recording to describe your issue?

benisbell, 2025-05-06:
https://github.com/user-attachments/assets/4003bf29-f3b1-4430-a19e-7bf9a637910a

BoBoYu-BamBu, 2025-05-07:
@benisbell  Oh, did you enable the 'Developer Mode' option in the preference settings? Once you enable the advanced options, you won't be able to set it anymore.
[image]

closed 2025-05-09 by benisbell