bambulab/BambuStudio #6828: 3MF files from V1 with STEP imported do not load in V2.0

technerd-official, 2025-05-05:
### Bambu Studio Version

Version 2.00.03.54

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 and Mac OS

### Printer

Bambu Lab P1S

### How to reproduce

If you try to open a 3MF file (saved in Bambu Studio V1) that had an imported STEP file it gives out an error - no geometry etc. in Bambu Studio V2.0.

### Actual results

3MF file saved in V1 won't open in Bambu Studio V2.0

### Expected results

Solution is to avoid importing a STEP file and instead import a STL file and save the 3MF file. This way the V1 3MF file works perfectly well in Bambu Studio V2.0

### Project file & Debug log uploads

[test_file.3mf.zip](https://github.com/user-attachments/files/20039407/test_file.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2025-05-06:
Hello, can you provide the original step file

jkenneydaniel, 2025-05-06:
I'm not sure if it's related, but I've noticed when attempting to import a STEP file via BambuStudio CLI - I receive an error similar to this one. Previously, STEP files could be imported no problem - but CLI seems to no longer support STEP.

Haidiye00, 2025-07-17:
https://github.com/bambulab/BambuStudio/releases
Hello, please use the latest beta version. Thank you~
If you have any further questions, we will analyze again

closed 2025-08-04 by Haidiye00