bambulab/BambuStudio #6831: 铺满打印板功能会生成出界模型

moe001, 2025-05-05:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

无关

### Additional system information

无关

### Printer

p1s

### How to reproduce

1. 打开附件工程
2. 右击热床中间的白色模型，选择铺满打印板

### Actual results

部分模型超出热床范围，无法进行切片。

[image]

### Expected results

正确铺满，不生成出界的模型

### Project file & Debug log uploads

[a.zip](https://github.com/user-attachments/files/20041294/a.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

ShanChang-BBL, 2025-05-12:
该问题已在内部版本解决，请耐心等待后续版本

fahaaaxiki, 2025-07-21:
Hello, we have solved the problem at latest release. please check [v02.02.00.60](https://github.com/bambulab/BambuStudio/releases/tag/v02.02.00.60)