bambulab/BambuStudio #6832: Latest Bambu update version ending in 89

CityGirl70, 2025-05-05:
### Bambu Studio Version

1.10.0.89

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10Pro 

### Additional system information

Intel(R) Core(TM) i5-7300U CPU @ 2.60GHz   2.71 GHz
64-bit operating system, x64-based processor

### Printer

Bambu Lab X1C

### How to reproduce

It says loading to printer but does not do anything after showing 100% file sent to printer

### Actual results

NOthing happens

### Expected results

I expected the printer to show the print file and start printing after slicing and sending

### Project file & Debug log uploads

I dont know how to zip file this debug log to you. I would just like to revert back to the old version. I didnt have any problems with it. I would not have updated but a file on Makerworld said I needed to , to be able to print it. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-BamBu, 2025-05-06:
@CityGirl70  Sorry, could you try another network instead?

CityGirl70, 2025-05-08:
What network do you suggest I try? I only have one network at home, which is where my printer is. All was well before the update. Are you one of the tech support, because honestly I feel your post was anything but helpful. I would like to roll this version back. Is that an option?