bambulab/BambuStudio #6833: Unkown Error 0500-0600-0002-00a1 after Studio and Firmware update on X1C

kdesroch1, 2025-05-05:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

HP Omen

### Printer

Bambu X1C

### How to reproduce

Turn the printer on, get this in the app and cannot print.  Doesn't show up in Bambu Studio

### Actual results

Unable to print!

### Expected results

To be able to print!

### Project file & Debug log uploads

Any project.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

shuizhiguo, 2025-05-07:
**Please upload log through Handy APP, and give us your ticket number**
***
***Upload log files through Bambu Handy***
     Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
     When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
     To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation.
     [image]

vampiro2004, 2025-05-09:
I have also just this error 05000600000200a1 with latest firmware and there’s no way to clear it

kdesroch1, 2025-05-09:
I ended up doing a factory reset, then unplugging power for a minute, then powering it back on.

Try just doing the power cycling first.