bambulab/BambuStudio #6834: New "Not support printing 2 or more TPU filaments" error when combining multiple TPU since BambuStudio 2.0

M-A-B007, 2025-05-05:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

X1C

### How to reproduce

1. Update BambuStudio to 2.0.* (before it worked perfectly fine)
2. Have a model combining two colors of TPU. 
--  It is supposed to print without AMS. The filament change can be done manually from an external spool while the print is paused. 
-- The pause can be invoked by pausing at a layer height (which is a valid feature in Studio UI at the "Preview" screen). 
-- Alternatively, the pause can be introduced directly in the custom-modified G-code. (Which is a valid feature on the "Printer settings" or "Filament settings" dialogues.)
3.  Switch from "Prepare" page to "Preview" page.

### Actual results

- When switching to "Preview" page, I immediately got the red error "Not support printing 2 or more TPU filaments."
- The error prevents me from continuing and inserting the pauses at layer height.
- I'm not able to convince the Studio, that I'm willing to manually change the filaments and don't want to use AMS.

Higlight:
This project worked perfectly fine on the prior versions of Bambu Studio. 
The error is superfluous, the printer hardware is perfectly capable of handling this case, it's only an artificial and stupid limitation introduced in the software. 

Suggestion:
Make a soft WARNING instead of hard ERROR out of this. Or introduce some property setting to suppress this error.

Overall outcome:
I'm going to throw the Bambu printers in the trash, and I'm going to buy the 5tup1d PRUSA again.
Really, this new behaviour is blocking the important projects in my workshop and is a big NO-GO for me at this moment.  It literally forces me to immediately buy other printers from concurrent manufacturers and never use Bambu again. Not happy, guys.
 

### Expected results

- After switching to "Preview" page, I can continue preparing the print. For example, I want to add the pauses at the layer height, so I can manually change the filament without AMS.  
- I would be perfectly ok if I got a soft WARNING that AMS is not able to support the changes of the desired filament materials...

### Project file & Debug log uploads

[MA3D-logo-QRsharpy-TPU30Dok-copy.zip](https://github.com/user-attachments/files/20050952/MA3D-logo-QRsharpy-TPU30Dok-copy.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

spuder, 2025-05-28:
The breaking change was introduced here: https://github.com/bambulab/BambuStudio/commit/60cdf3b6551a8c18c10db0a746e1b15b764eda66

dsmaher, 2025-05-31:
I did some more poking around and it _looks_ like the check has been removed... but not released yet: [cd32193](https://github.com/bambulab/BambuStudio/commit/cd3219e6654d126363c04d0135bece62dcf94e39).  So hopefully next release? 🤞

XunZhangBambu, 2025-06-03:
@dsmaher @spuder @M-A-B007  Hi everyone, now we support slice with 2 or more tpu filaments with out latest version 02.01.00.59