bambulab/BambuStudio #6836: CLI: Unable to load process.json correctly

arirangz, 2025-05-06:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windonws 10 pro

### Printer

Bambu LAB A1 Mini

### How to reproduce

Hello,

I am using the following proccess.json:
{ "type": "process", "brim_type": "outer_only", "enable_support": "1", "from": "User", "inherits": "0.20mm Standard @BBL A1M", "initial_layer_infill_speed": "100", "initial_layer_speed": "30", "name": "0.20mm Standard @BBL A1M - PeTG Round shape", "print_settings_id": "0.20mm Standard @BBL A1M - PeTG Round shape", "skirt_loops": "1", "sparse_infill_density": "75%", "top_surface_pattern": "concentric", "version": "1.8.0.25", "wall_loops": "7", "compatible_printers": [ "Bambu Lab A1 mini 0.4 nozzle" ] } 

When I open the generated 3MF, the sparse_infill_density is not 75% but the default 15% value. So the overridden value are not being taken.
See this issue:
https://github.com/bambulab/BambuStudio/issues/1963

How can I fix this?

Someone said that we have to export the full settings but when I export the process, I am only getting the short version with inherits.

Thanks

### Actual results

When opening the 3MF, I don't see the overridden settings from the process.json but only the default values from parent profile.

### Expected results

All settings should be loaded

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

gommzystudio, 2025-05-19:
I found another parameter that seems to be ignored by the CLI, this time in machine.json:
`"machine_max_acceleration_x": [
  "12000",
  "12000"
],
"machine_max_acceleration_y": [
  "12000",
  "12000"
]`

However, the generated G-code contains:
`; machine_max_acceleration_x = 6000,6000
; machine_max_acceleration_y = 6000,6000`

This works correctly when slicing in the GUI version of Bambu Studio, so it seems like the CLI is not applying these values properly.

I noticed this because the total print time differed by around 5% between the GUI and CLI-generated G-code.

gommzystudio, 2025-05-19:
I did some more digging and found that this behavior is actually caused by the CLI’s safety mechanism.

The CLI explicitly checks for safe parameter limits using the printer’s configuration and clamps any values that exceed them. So even though "machine_max_acceleration_x" and similar settings are defined in machine.json, the CLI enforces a stricter limit internally — in my case, falling back to 6000,6000. This explains why the GUI uses 12000 while the CLI doesn’t: the safety check is only applied in the CLI, not in the GUI.

It’s a bit unfortunate that there’s currently no flag or override to disable this safety check — especially for advanced users who know what they’re doing — but I understand that this was likely intentional to protect the hardware in automated CLI environments.

So in the end, it’s not a bug, but it would be great if a --disable-safety-checks option existed for power users.

arirangz, 2025-05-19:
@gommzystudio Thanks for your feedback and for digging into it.

So do you think   "sparse_infill_density": "75%" is also beyond the safety limit?

But from my test,   "top_surface_pattern": "concentric", is also not being taken and I don't think this should cause any issue.

Thanks

gommzystudio, 2025-05-19:
I think that's not caused by the safe mode, because it only overwrites printer settings.

arirangz, 2025-05-22:
after digging, I understand that the CLI is not managing inherited settings.

So we have to add the full settings in each json.

When I am trying to export settings from the export menu, I only get json with custom values and refrence to parent profile.

Is it still possible to export full json?

I check in C:\Program Files\Bambu Studio\resources\profiles\BBL\process and file inherit from another one so we can't get a full settings file. Is there a solution for that?

By the way, why CLI is not handling inherits? From what I read it's not even a good solution to have parents settings hard coded in the json as those settings may be updated by Bambu updates.

@FiredEgg-bambu any help on this?

Thanks

lanewei120, 2025-05-23:
> after digging, I understand that the CLI is not managing inherited settings.
>
> So we have to add the full settings in each json.
>
> When I am trying to export settings from the export menu, I only get [...]

Hi @arirangz 

CLI has limited logic to process the jsons, we suppose the full json has been generated before when passed to CLI

maybe we can add some logic to export full json in bambu studio

lanewei120, 2025-05-24:
I don't have the python script currently 
however i think it is easy to ask chat-gpt to do it

for example:
`import os
import json
from typing import Dict

def index_json_files(root_dir: str) -> Dict[str, str]:
    json_map = {}
    for dirpath, _, filenames in os.walk(root_dir):
        for filename in filenames:
            if filename.endswith(".json"):
                name_without_ext = filename[:-5]
                json_map[name_without_ext] = os.path.join(dirpath, filename)
    return json_map

def load_full_json(name: str, json_map: Dict[str, str], visited=None) -> dict:
    if visited is None:
        visited = set()
    if name in visited:
        raise ValueError(f"Circular inheritance detected: {name}")
    visited.add(name)

    if name not in json_map:
        raise FileNotFoundError(f"'{name}.json' not found in scanned directories")

    with open(json_map[name], "r", encoding="utf-8") as f:
        data = json.load(f)

    base_data = {}
    if "inherits" in data:
        parent_name = data["inherits"]
        base_data = load_full_json(parent_name, json_map, visited)

    merged = {**base_data, **data}
    return merged

if __name__ == "__main__":
    import argparse

    parser = argparse.ArgumentParser(description="Resolve JSON inheritance")
    parser.add_argument("json_name", help="JSON file name without .json (e.g. my_config)")
    parser.add_argument("--root", default=".", help="Root directory to scan")
    args = parser.parse_args()

    json_files = index_json_files(args.root)
    resolved = load_full_json(args.json_name, json_files)

    print(json.dumps(resolved, indent=2, ensure_ascii=False))
`

save the above codes into run.py, and run as follows:
python run.py "Bambu PLA Sparkle @BBL H2D" --root resources/profiles/BBL/filament

the full json will be output on the screen

lanewei120, 2025-06-01:
Hi @arirangz 

I am sorry we don't have this plan currently

lanewei120, 2025-06-04:
> Hi [@lanewei120](https://github.com/lanewei120)
> 
> Then, do you think you can improve CLI to automatically get parent JSON?
> 
> Thanks

Hi @arirangz 

we also don't have such plans

thanks