bambulab/BambuStudio #6842: 最新版BambuStudio无法打开，报错unable to automatically switch to MESA OpenGL library.

shimodianji, 2025-05-06:
### Bambu Studio Version

v02.00.03.54

### Where is the application from?

Bambu Lab github releases

### OS version

Win10

### Additional system information

[image]

### Printer

A1mini

### How to reproduce

无

### Actual results

无

### Expected results

无

### Project file & Debug log uploads

无

### Checklist of files to include

- [ ] Log file
- [ ] Project file

guanyun-gudujian, 2025-05-07:
@shimodianji   log file pls

behold81, 2025-05-20:
I have this same issue. it was working on 2.02 and now even with a downgrade it fails to work. I can provide logs but need to know where from

closed 2025-05-20 by shimodianji

behold81, 2025-05-20:
No that didn't work. I have been doing that to try and fix it.  

Now you have closed the issue and I will have to create a new one.

GuDuJian-J-Zhang, 2025-05-21:
> I have this same issue. it was working on 2.02 and now even with a downgrade it fails to work. I can provide logs but need to know where from
here:
[image]

guanyun-gudujian, 2025-05-21:
> I have this same issue. it was working on 2.02 and now even with a downgrade it fails to work. I can provide logs but need to know where from

here:
[image]

behold81, 2025-05-21:
> > I have this same issue. it was working on 2.02 and now even with a downgrade it fails to work. I can provide logs but need to know where from
> 
> here:
> [image]

Where is that location as it never loads that far it errors on starting.

guanyun-gudujian, 2025-05-21:
> > > I have this same issue. it was working on 2.02 and now even with a downgrade it fails to work. I can provide logs but need to know where from
> > 
> > 
> > here:
> > [image]
> 
> Where is that location as it never loads that far it errors on starting.

sorry, my stupid

[image]