bambulab/BambuStudio #6843: Support Bottom interface Layers available option in search, but is hidden in UX list by mistake

psiberfunk, 2025-05-06:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS

### Additional system information

N/A

### Printer

H2D

### How to reproduce

1. Go to Support settings in any profile.
2. Search for "interface layers" --> see that bottom interface layers appears (is an option in Orca and elsewhere) in search: [image]
3. Go look at the ACTUAL list of options you can change... you cannot find Bottom Interface layers anywhere: [image]

### Actual results

Can't find bottom interface layers setting, but I cannot because it's hidden.

### Expected results

Bottom interface layers setting should be exposed in UX.

**I want to change the bottom interface layers setting, but I cannot because it's hidden.**

### Project file & Debug log uploads

N/A.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

NothFie, 2025-05-07:
@psiberfunk Thanks for your feedback. In Studio 2.0, we have changed strategy about bottom interface, and this parameter is removed from Tree Support but still available in Normal Support. If you still want to use bottom interface in Tree Support, would you please tell me your situation so that I can provide you with a targeted strategy.

NothFie, 2025-05-07:
@psiberfunk I got your thoughts. But with many tests on H2D, using support material as bottom interface failed a lot because bottom interface surfaces with small areas can easily fall out of the model while printing. And printing with no bottom interface but 0.2 bottom_z with the same material as model makes it easy to remove supports while also having stronger stickiness than bottom support materials. Have you tried set bottom_z larger like 0.3? This might also help you get the clean model.

psiberfunk, 2025-05-07:
I know many of these tricks and problems from over a decade of printing .  It’s fine to have it not be the default , but there are situations where this is **needed** .  One clear example is a model where normally I would use traditional supports for one part of it and tree supports for another but because I need tree supports and I cannot mix support types, I need to use tree supports.  However , I want clean interfaces , and I want to use “sticky” main model materials like TPU or PETG.  These materials almost always mark model surfaces if I use them “not only on buildplate” .  Larger Z gaps here tend to either be too sticky or not bind reliably enough.  Using a different interface material is a prime use case for this situation and the H2D and other printers that can switch filaments are perfect for this .  If you’re concerned about print failures , don’t make it the default , or warn the user about the danger if you must.

The problem I have here is you took away ** the option to do what I want, even if it may not work**.  3D printing users need to be able to tweak and experiment away from the default settings … taking away that freedom to innovate and experiment is soul crushing and frustrating for people who have specific needs.   Just because you cannot imagine a use case where it is needed or is not done better in a different does not mean it does not exist .  Please don’t take that as an insult , but rather a clear request to let people tinker :  it’s needed to produce the best models and results . 

 If you want the best creators to use your tools .. don’t completely gimp the tools so we can’t realize our visions with them :D.

NothFie, 2025-05-08:
@psiberfunk understood, sorry for the bad experience you had and we'll discuss about it.

psiberfunk, 2025-05-08:
> [@psiberfunk](https://github.com/psiberfunk) understood, sorry for the bad experience you had and we'll discuss about it.

Thanks for considering it.  I understand you want to keep the software accessible, and you've got optimization reasons for doing things this way.. but it's difficult to understand what's going on when you **swear that option was there** and then suddenly "poof" it's gone , but not in the search !  It makes you think you're crazy :)

psiberfunk, 2025-08-17:
Yup.  Also want this.

LexAKABox, 2026-05-07:
if this thing didnt work so dang well.. i would have moved on the minute they started forcing all prints to go through their servers unless you give up all online and mobile features.  this is likely for continued Chinese IP theft... but the corrupt US government will use it for more nefarious reasons and i highly doubt they will spend the time and money to fight back.

alikadir, 2026-09-09:
@NothFie  Instead of completely hiding this feature, the user should be warned about any potential issues when they attempt to use it. This way, the user is made aware of the possible problems, but if they still want to use the feature, they should be allowed to proceed.

LexAKABox, 2026-09-09:
> [@NothFie](https://github.com/NothFie) Instead of completely hiding this feature, the user should be warned about any potential issues when they attempt to use it. This way, the user is made aware [...]

That could be acceptable. As long as its restored... heck they still have grid as the default infill.. im sure that causes people way more issues than the ability to use a support interface

aidanschmidt-maker, 2026-09-11:
I really need this option, crazy to just block it