bambulab/BambuStudio #6845: H2D slicing uses way too much interface material

cmontis, 2025-05-06:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Pro Version	10.0.26100 Build 26100

### Printer

Bambu Lab H2D

### How to reproduce

Simply load the file.
Go to the support tab and make sure that support base is default and support interface is PVA. and slice it.
In the slicing results change drop down to filament and see that we are using 34.55m of interfacing!

This is a dual nozzle design and there is no way that a small amount of interfacing support material is using 34.55m of material.  Something is not right here....  We have two nozzles.

[image]

### Actual results

the result is nozzle 1 PETG-HF uses 253.91m and nozzle 2 PVA is using 34.55m for just the support interface (no tower is enabled.

### Expected results

I expected a minimum amount of interfacing material to be used since I was not using it for the support base.  Just the support interface is a tiny amount of material.

I am seeing this issue when taking files form Maker world and they are defaulted for X1C.  I click H2D download but it defaults to X1C.  When I select my H2D and synch the printer it seems something is not crossing over.  I think this is where the issue is happening.

### Project file & Debug log uploads

[h2d ams 2 dry dispencer.zip](https://github.com/user-attachments/files/20072402/h2d.ams.2.dry.dispencer.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

NothFie, 2025-05-07:
@cmontis Thanks for your feedback. I opened your 3mf and found it going well. As you suspected, it might have been processed using X1C logic while slicing for some reason. Saving and reopening the file may resolve the issue. We’ll continue investigating the cause of such occurrences.

[image]

hotellonely, 2025-05-11:
> [@cmontis](https://github.com/cmontis) Thanks for your feedback. I opened your 3mf and found it going well. As you suspected, it might have been processed using X1C logic while slicing for some [...]

would it be possible to limit the range of interface expansion? in my model i'm having issues with the interface, because it's has surrounded my thin rod. In such case, it's very hard to remove the support&interface without breaking the part.

NothFie, 2025-05-12:
@hotellonely It can be limited by setting the parameter "Normal Support Expansion" with a value >0. When it's 0, it means auto and the range of expansion is set to 2mm.

hotellonely, 2025-05-12:
> [@hotellonely](https://github.com/hotellonely) It can be limited by setting the parameter "Normal Support Expansion" with a value >0. When it's 0, it means auto and the range of expansion is set to 2mm.

I think you were talking about the base but I was talking about the top interface. It doesn't work for the top interface overgrowth.

NothFie, 2025-05-12:
@hotellonely I'm a bit confused, could you please provide some pictures to help me understand?

NothFie, 2025-05-12:
@hotellonely It is possible that you're meaning Normal support with Grid(Default) style and what I said above meant to help Tree support with Hybrid(Default when top_z is 0) style with top_z set to 0. If so, you could reduce the "Base pattern spacing" to reduce the Normal support interface range, or simply use Snug style.

cmontis, 2025-05-12:
> [@hotellonely](https://github.com/hotellonely) I'm a bit confused, could you please provide some pictures to help me understand?

The photo is included.  It demonstrates a very large amount (almost 13m) of material was calculated for l support.  When I select default for interface and then pva for support interface....only a small amount should be calculated.  I figured out the slicer error by changing from default to my main filament and it corrected.  It did not use to require me to list anything specifically (i.e. that is what default should do...).

cmontis, 2025-05-12:
> [@hotellonely](https://github.com/hotellonely) I'm a bit confused, could you please provide some pictures to help me understand?

Photo is in the post.

NothFie, 2025-05-13:
@cmontis What I said above is talking to another one who asked about the expansion of the interface. As for your problem, it's indeed a BUG that when you did NOT enable the WIPE TOWER, the option "avoid interface filament for base" is not working. We're looking into it and will inform you as soon as it's fixed.