bambulab/BambuStudio #6846: Bambu studio fails to print because it thinks I'm using a different plate than I have selected

ssylvan, 2025-05-07:
### Bambu Studio Version

2.0.3.54

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

Note that I have the textured plate selected, but I can't even slice the model because Bambu studio somehow thinks I have the cool plate selected. 

[image]

### Printer

X1C

### How to reproduce

Load https://makerworld.com/en/models/1131837-1-inch-25mm-strap-keeper?from=search#profileId-1132233
Select a PETG HF filament from AMS. 
Try to slice. 

### Actual results

See screenshot. BS refuses to slice because it thinks I'm using the cool plate when I'm not (and textured PEI is selected). 

### Expected results

It should read the actual plate setting correctly and not refuse to slice. 

### Project file & Debug log uploads

[Bag Strap Keeper(2).zip](https://github.com/user-attachments/files/20075807/Bag.Strap.Keeper.2.zip)

[debug_network_Tue_May_06_20_46_48.log.zip](https://github.com/user-attachments/files/20075815/debug_network_Tue_May_06_20_46_48.log.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

OzButcher78, 2025-05-11:
I have the same issue, reported on the bambu forums. The problem is when using multiple plate project and manually assigning plates different to global. Bambu Studio ignores individual plate settings and only reads global. 

[image]

OzButcher78, 2025-05-11:
> 
> ### Additional system information
> Note that I have the textured plate selected, but I can't even slice the model because Bambu studio somehow thinks I have the cool plate selected.
> 

You actually do have the ''cool plate'' selected, it's defined when clicking on that hexagon icon next to the build plate. This tool is useful when printing with multiple build plates where some need to be different typess. 

However, a issue remains with Bambu Studio that it will only slices with the Global Build plate, so bug still exists.