bambulab/BambuStudio #6847: Since Update today

MartinoSwiss, 2025-05-07:
### Bambu Studio Version

Bambu Lab X1 Carbon 01.09.00.01 / AMS 00.01.06.62

### Where is the application from?

Bambu Lab Official website

### OS version

Win 11

### Additional system information

Pentium 13th Gen i9-13900K
ROG Maximus Z790 HERO
Microsoft Windows 11 PRO
2x 16GB RAM 6000MHz Samsung 990 PRO
MSI Nvidia RTX 4090 Gaming X Trio 24GB

### Printer

Bambu Lab X+ Carbon

### How to reproduce

Always by slice, all print data, directly crash to Desktop, without report

### Actual results

Always by slice, all print data, directly crash to Desktop, without report

### Expected results

Bambu LAb Studio crash directly

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/20080426/log.zip)

It doesn't require any "project files" because this problem now applies to ALL print media that I just printed.

### Checklist of files to include

- [x] Log file
- [x] Project file

BoBoYu-BamBu, 2025-05-07:
@MartinoSwiss  Sorry, could you please provide the 3mf file?

MartinoSwiss, 2025-05-07:
Hello Bambu Support

I can send ALL i have, same problem, since Patch today

Best Regards
Martin Obeerhänsli

MartinoSwiss, 2025-05-08:
Hallo Bambu Team

Gibt es schon Neuigkeiten? Ich würde gerne wieder mal drucken.
Oder wie kann ich das Update/Patch wieder zurücksetzen?

Freundliche Grüsse
Martin Oberhänsli

BoBoYu-BamBu, 2025-05-08:
@MartinoSwiss  Sorry, could you try reinstalling Studio in a brand new environment? Delete all the configuration folders.

MartinoSwiss, 2025-05-08:
Dies hat leider auch kein Erfolg gebracht.
Beim Start hatte ich folgende Meldung

BoBoYu-BamBu, 2025-05-08:
@MartinoSwiss Could you please provide us with your 3mf file so that we can reproduce the result and solve the problem?

MartinoSwiss, 2025-05-08:
you can get any 3mf file you find from Makerlab

MartinoSwiss, 2025-05-09:
Das war die Lösung dank eines Bambu lab Forum Deutschland, trotzdem denke
ich, müsst ihr hierbei etwas unternehemen.

-STRG+ALT+ENT drücken und den Taskmanager öffnen

-in den Reiter "Details" wechseln

-Bambu-studio <https://www.bambu-forum.org/wiki/entry/28-studio/?synonym=68>.exe
suchen, rechtsklicken und "zugehörigkeit festlegen" auswählen

-Den Haken bei "alle prozessoren" rausnehmen und stattdessen nur "ein paar"
auswählen - die hälfe zum Beispiel.

JohnSmithCYM, 2025-05-12:
Thanks for your feedback and can you tell us where the forum give a solution to this problem? We want to make clear of this bug.

JohnSmithCYM, 2025-05-27:
https://github.com/Haidiye00/BambuStudio/actions/runs/15222819657/artifacts/3189644769
@Haidiye00 have compiled a temporary package here. Can you download it and help me see if it can slice without crashing~thanks